sipeed/MaixPy-v1_scripts #1: Custom model for Object Detection

Ace95, 2019-04-04:
Hi, I'd like to know how to create my custom object detector model. I tested the "face-detector" with a different tiny YOLOv2 model but it didn't work. Any repo I can check to find more infos about?

Thanks!

PTS93, 2019-04-30:
Yes, running full yolo or tinyYolo would be interesting.

Ace95, 2019-06-13:
> Yes, running full yolo or tinyYolo would be interesting.

They update the Sipeed blog with a guide to train,convert and run MobileNet on the MaixPy board. 

http://blog.sipeed.com/p/680.html#more-680

Also here a nice guide for transfer learning. I tested it and it worked out fine, I may upload my full repo with description next week.

https://www.instructables.com/id/Transfer-Learning-With-Sipeed-MaiX-and-Arduino-IDE/
https://www.youtube.com/watch?v=fZ4QbtAjKXo&t=30s

AIWintermuteAI, 2019-11-13:
https://www.instructables.com/id/Object-Detection-With-Sipeed-MaiX-BoardsKendryte-K/
Here's my second article and video, specifically about training custom object detection models for K210. Cheers!

Ace95, 2019-11-22:
Thanks @AIWintermuteAI, I'll try to implement the object detector as well. I'm collecting some works about Transfer Learning, both developed with TF/Keras (Python) and in ML.NET(C#), in this repo:

https://github.com/Ace95/transferlearning_AIDay

Feel free to comment and expand the repo with more examples.