sipeed/MaixPy-v1_scripts #2: speaker sample

mhama, 2019-04-08:
It would be better to have some samples to use the speaker, as Maix Go have a good-looking speaker hardware.

Neutree, 2019-05-09:
https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/audio/play_wav.py

linycm, 2019-05-11:
I test play_wav.py and got the below error. Pls advise. thanks!

wav_info = player.play_process(wav_dev)

[MAIXPY]: wav error code : 7
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: wav init error

And for player.volume(40), I got error:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'AUDIO' object has no attribute 'volume'

Neutree, 2019-05-11:
@linycm wait for the new release or build yourself firmware

linycm, 2019-05-12:
@Neutree I build firmware again but still got error when
wav_info = player.play_process(wav_dev)
[MAIXPY]: wav error code : 7
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: wav init error
--------
core dump: fault store
Cause 0x0000000000000007, EPC 0x00000000800c2a44
.......

Neutree, 2019-05-13:
@linycm The header of wav file you used have some problem, please send me your wav file, so I can checke the header, or just convert to wav file maixpy support. [here](https://cloudconvert.com/mp3-to-wav) for example

linycm, 2019-05-21:
[radio.wav.tar.gz](https://github.com/sipeed/MaixPy_scripts/files/3202723/radio.wav.tar.gz)

@Neutree pls help to check. thanks!

Neutree, 2019-05-22:
@linycm This audio file uses ADPCM encoding. MaixPy only supports PCM 16bit little endian encoding. Please find a way to convert to such a format. There is no plan to support ADPCM.

[image]

MaixPy support like this one:

[image]