sipeed/MaixPy-v1_scripts #3: error when load model

kimchicanh, 2019-05-17:
Hi,
I am just a newbie
I want to run face detection but I get error when I run :
    $  task = kpu.load(0x00300000) 
ERROR:                                                
v=596458738, flag=-2096411809, arch=-1955282835, layer len=834782759, mem=-39249
0997, out cnt=-663129117                                                        
err: we only support V3 now, get V596458738                                     
[MAIXPY]kpu: kpu_model_get_size  error -3Traceback (most recent call last):     
  File "<stdin>", line 1, in <module>                                           
ValueError: [MAIXPY]kpu: load error -2  

please help me
thanks

Neutree, 2019-05-17:
https://github.com/sipeed/MaixPy/issues/82
model: http://dl.sipeed.com/MAIX/MaixPy/model/face_model_at_0x300000.kfpkg
burn tool: https://github.com/sipeed/kflash_gui
pack `*,kfpkg` file : http://blog.sipeed.com/p/390.html

kimchicanh, 2019-05-17:
@Neutree 
I get this error when I run kflash on ubuntu 16.04:

kflash_gui_v1.0.0_linux/libpython3.6m.so.1.0': dlopen: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found (required by /home/kflash_gui_v1.0.0_linux/libpython3.6m.so.1.0)

do you know how to use command line with kflush

Thanks

Neutree, 2019-05-17:
sipeed/kflash_gui#1