sipeed/MaixPy-v1_scripts #4: put 2 model to sipeed

kimchicanh, 2019-05-20:
Hi,

I want to use 2 model, ex 1MB vs 3 MB, how can I put it to sipeed and read it

Thanks

Neutree, 2019-05-20:
[pack them to `*.kfpkg` file](http://blog.sipeed.com/p/390.html), and download to flash with [kflash](https://github.com/sipeed/kflash_gui)

kimchicanh, 2019-05-20:
@Neutree thanks you,
What about kmodel, how to create it

Thanks

Neutree, 2019-05-20:
maybe these links can help you?

http://blog.sipeed.com/p/category/maix-software

https://www.youtube.com/watch?v=fZ4QbtAjKXo&t=30s

https://www.instructables.com/id/Transfer-Learning-With-Sipeed-MaiX-and-Arduino-IDE/

kimchicanh, 2019-06-03:
@Neutree 
Thanks for your help,
I create 2 kmodel, first have 388776 (0x5EEA8) bytes so I put model at 0x00300000 and 0x0035EEA9 but when I use kpu to load model, it cannot load model 2:
task = kpu.load(0x300000)
task_re = kpu.load(0x0035EEA9)

thanks