sipeed/MaixPy-v1_scripts #5: Issue running demo_net_socket.py script

marcstringer, 2019-05-24:
File "<stdin>", line 1, in <module>
  File "/flash/demo_net_socket.py", line 10, in <module>
AttributeError: type object 'board_info' has no attribute 'WIFI_RX'

Getting the above error.
Running the demo script 'demo_net_socket.py ' I have changed the nic,connect() with my own wifi ssid and password.

Any help here would be appreciated

Neutree, 2019-05-24:
firmware version?

marcstringer, 2019-05-24:
MicroPython v0.1.1-96-g71e5beaa7-dirty on 2019-03-08; Sipeed_M1 with kendryte-k2                                                                                                   10

Not sure how to check i get the above line on serial conneciton

Neutree, 2019-05-24:
upgrade to the [latest firmware](https://github.com/sipeed/MaixPy/releases/download/v0.3.2/maixpy_v0.3.2_no_lvgl.bin) first with tool [kflash_gui](https://github.com/sipeed/kflash_gui)

or see [upgrade doc](https://maixpy.sipeed.com/zh/get_started/upgrade_firmware.html#)(use google translate plugin)

or see [video](https://youtu.be/K46kFRur-Hw?t=296)(select english subtitle)

TakuroFukamizu, 2019-07-23:
Dear @Neutree san.
AttributeError is fixed at the latest firmware. 
But, I got a new error.

[image]