sipeed/MaixPy-v1_scripts #6: touchscreen on GO

machinehistories, 2019-05-29:
When I run the touchscreen sample I just get a blank screen and in the terminal I get lines of 
0 0 0 
0 0 0
If I use the thumbwheel I get 
{"w":320, "h":240, "type"="rgb565", "size":153600}
0 0 0. 
I noticed in the FFT demo there are alternative pins called out. Might this also be the case for the touchscreen. Do you have any other suggestions. thanks

Neutree, 2019-05-29:
Specify pins here: https://github.com/sipeed/MaixPy_scripts/blob/639e9dded8139b51f1e2457406462e8cde9e0bdc/hardware/demo_touchscreen.py#L11

if you use go, that should be like this:

[image]
[image]

machinehistories, 2019-05-29:
Those are the pins I used but unfortunately the touchscreen is
unresponsive. I was really excited to use it too. Do you have any other
suggestions or is this something I should take up with Seeed where I
purchased this product from. - jason

Neutree, 2019-05-30:
the touchscreen device is `ns2009`, communicate with i2c, you can use `i2c.scan()` to find device:
```
from machine import I2C
i2c=  I2C(I2C.I2C0, freq=200000, scl=30, sda=31)
devices = i2c.scan()
print(devices)
```
output:
```
[38, 72]
```

if no device found, maybe the hardware broken or connection broken, maybe you have to  check the connection by multimeter!

machinehistories, 2019-05-31:
When I run

from machine import I2C
i2c=  I2C(I2C.I2C0, freq=200000, scl=30, sda=31)
devices = i2c.scan()
print(devices)

I do get output:

[38, 72]

However it continues to only output 0 0 0 no matter how or where I touch
it. I checked the continuity from pins 30,31 to the chip and they were ok.
I also checked continuity from the lcd connector pins 30, 31 to the
mainboard pins and those had continuity. The connection to the lcd board to
the ribbon cable looks secure and I have reseated it but it didn't change
anything. I think it is something in the touch sensor. Are there any
documents for the lcd board and with pinouts for the flat cables and is
there anything I should check on the lcd board to see if the sensor is
broken. thanks

Neutree, 2019-06-01:
http://dl.sipeed.com/MAIX/HDK/Maix-GO/Maix-GO_2.20/Maix_GO_2.20%28Schematic%29.pdf
http://dl.sipeed.com/MAIX/HDK/Maix-LCD/Maix_LCD_12.07/Maix_LCD_12.07%28Schematic%29.pdf

https://github.com/sipeed/LicheeDan_K210_examples/tree/master/src/lvgl

machinehistories, 2019-06-01:
Here is what I discovered. I also tried it using the arduino ide and I get
the same result. The i2c address scans, the lcd works, however I get no
feedback on the touchscreen. I appreciate your help. Do you suggest I get
another screen, should I replace the entire GO kit, or are there other
tests you can suggest.
[image: lcd.PNG]

xelll, 2019-06-20:
you can replug the lcd module, maybe the connect is not so good.

xelll, 2019-06-20:
you can use this firmware to test touchscreen. [ns2009.bin](https://fdvad021asfd8q.oss-cn-hangzhou.aliyuncs.com/Sipeed_M1/firmware/ns2009.bin)

machinehistories, 2019-06-21:
I appreciate your suggestions. Unfortunately it still doesn't respond to
touch. I am going to get a replacement board and I look forward to using
it.

xelll, 2019-06-22:
ok...