sipeed/MaixPy-v1_scripts #8: AttributeError: 'ESP8285' object has no attribute 'scan'

TakuroFukamizu, 2019-07-19:
I've tried to run below script at my Sipeed MAix Dan Dock(M1w model). 
https://github.com/sipeed/MaixPy_scripts/blob/master/network/demo_wifi_ap_scan.py

but got the following error.
```
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
AttributeError: 'ESP8285' object has no attribute 'scan'
```

What's wrong?

- Board: MAix Dan Dock
- MicroPython 59c658e-dirty on 2019-04-26; Sipeed_M1 with kendryte-k210
- Firmware: https://bbs.sipeed.com/t/topic/688

Neutree, 2019-07-20:
this version have no this attribute, but the 0.3.2 do

closed 2019-07-20 by Neutree

TakuroFukamizu, 2019-07-21:
Dear @Neutree 

It's not fixed yet. Because `nic.scan()`  return SSID **ONLY**. but this code expects SSID, RSSI, and MAC address.
https://github.com/sipeed/MaixPy_scripts/blob/6928fe172a77402915850a27c6f8fe04a75b8223/network/demo_wifi_ap_scan.py#L21

[image]