sipeed/MaixPy-v1_scripts #11: kpu.forward OS Error

Narduzzi, 2019-07-23:
Hello,

I'm trying to run the following code : 

```
import sensor
import image
import lcd
import KPU as kpu

lcd.init()
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.run(1)
task = kpu.load(0x300000) # you face model in flash at address 0x300000
print("Done")
img = sensor.snapshot()

a = kpu.forward(task, img,1)  #<--- this is the problematic line

print("OK")
```

However, MaixPy IDE return the message "OS ERROR". 
What could be the problem? How can I solve it?

Thank you for your help!

minyatos, 2020-05-31:
> a = kpu.forward(task, img)  #<--- this is the problematic line

Neutree, 2020-08-10:
upgrade firmware and try again https://dl.sipeed.com/MAIX/MaixPy/release/master

closed 2020-08-26 by Narduzzi