sipeed/MaixPy-v1_scripts #23: No module named cpufreq

Maximellerbach, 2019-11-30:
hey, I noticed when I tried to change/check the cpu/kpu frequency that the cpufreq module doesn't seems to be working, I installed the maixpy_v0.3.2_full version, I don't think I have made anything wrong. Hope this problem will be fixed soon !

`import cpufreq
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'cpufreq'`

Neutree, 2019-12-01:
upgrage firmware  http://dl.sipeed.com/MAIX/MaixPy/release/master/

closed 2019-12-01 by Neutree

Maximellerbach, 2019-12-01:
even with the latest  v0.4.0-101-g553fdfd it doesn't work !
[image]

Maximellerbach, 2019-12-01:
or with v0.5.0
[image]

Neutree, 2019-12-02:
https://maixpy.sipeed.com/zh/libs/Maix/freq.html

Maximellerbach, 2019-12-02:
thanks a lot !

Maximellerbach, 2019-12-02:
when tring to change kpu frequency (set function is working fine with cpu frequency):
[image]

Maximellerbach, 2019-12-02:
nevermind, I found where the problem was !
Thank you again !