sipeed/MaixPy-v1_scripts #27: after powering and the boot.py procedure how to execute a program?

ARMARATRON, 2020-01-16:
just as the title says, I just do not find a how to after the boot procedure execute a program, (in this case a simple menu in which you can select to run a file ("the mythical user defined script") with the:
////////////////(yes it should be # but....)
            with open("/sd/SCRIPTS/"+cadena[x]) as f:
                exec(f.read())
///////////////
my boot.py file:
`import image, time, lcd #uos , sensor, ujson,`
`from Maix import GPIO`
`from machine import Timer`
`print("ESTAS en:")`
`os.getcwd()`
`print("que tiene:")`
`os.listdir()`
`print("en SCRIPTS hay:")`
`os.listdir('/sd/SCRIPTS')`
`print("en IMGS hay:")`
`os.listdir('/sd/IMGS')`
`time.sleep(0.1)`
`#Tim=Timer(Timer.TIMER0, Timer.CHANNEL0, mode=Timer.MODE_PERIODIC, period=1, unit=Timer.UNIT_MS,  callback=on_timer, arg=on_timer, start=False, priority=1, div=0)`
`lcd.init(freq=680000, color=0x0000)`
`time.sleep(0.3)`
`relog=time.clock()`
`SWATT = image.Image(copy_to_fb=1)`
`#print(Tim)`
`#sys.version()`
`x=0`
`while x<1:`
`    x=x+1`
`    relog.tick()`
`    SWATT.draw_string(25, 100, "SWATT 2.0", (236,36,255), 4)#, 0, 0, False, 0, 0, 0, 0, 0, 0)`
`    lcd.display(SWATT)#9990 ROSA OBCURO, 0222 VERDE, 0555 verde+ 004;`
`    time.sleep(0.7)`
`    lcd.clear()`
`    SWATT.clear()`
`    SWATT.draw_string(75, 100, "SWATT 2.0", lcd.WHITE, 2)#, 0, 0, False)#, lcd.BLACK)`
`    lcd.display(SWATT)`
`    time.sleep(0.65)`
`    lcd.clear()`
`    SWATT.clear()`
`#import Main.py`
`with open('/sd/021menu.py') as f:#this does not works when booting, it works if I use the MaixpyIDE`
`    exec(f.read(),globals())#`

Neutree, 2020-08-07:
duplicate with #207

closed 2020-08-07 by Neutree

ARMARATRON, 2023-09-13:
#207 does not seem to exist, did you mean #27? 
If so #27 is this issue...