sipeed/MaixPy-v1_scripts #56: How to add Speech Reconizer demo to Maixpy??

GuilhermeAgostin, 2020-06-22:
Hi, I'm a beginner with sipeed maix boards, and I don't know how to add the demo Speech Reconizer in the Maixpy ​​IDE, can someone help me?

huoxingwen, 2020-06-23:
I have the same trouble

shintarof, 2020-06-26:
When you build the firmware, you can use it by selecting Enable Speech recognizer in the Modules configurations section

GuilhermeAgostin, 2020-06-27:
Olha, atualmente estou usando como exemplo esse codigo:  https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/speech_recognizer/speech_recognizer.py

Minha Versao de IDE do Maixpy é a 0.2.5 

[image]

E como arquivo bin 

[image]

Infelizmente ainda nao entendi o que fazer @shintarof

shintarof, 2020-06-28:
Please refer to the following instructions on how to build the firmware

https://github.com/sipeed/MaixPy/blob/ master/build.md

lancewj, 2020-07-16:
你好我想问下，你的运行成功了，我运行之后提示以下错误。
[image]
[image]
我把提示报错的相关部分注释后提示后，能够记录语音，但是不能识别。
请问有遇到过这种情况吗

shanejohnpaul, 2020-08-02:
I tried to run this demo. Right now there is no module named SpeechRecognizer.

Neutree, 2020-08-10:
@QinYUN575

junhuanchen, 2020-11-18:
https://github.com/sipeed/MaixPy_scripts/tree/master/multimedia/speech_recognizer

junhuanchen, 2020-11-18:
https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/speech_recognizer/isolated_word.py
https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/speech_recognizer/test_maix_asr.py
in maixpy_0.5.1_128 +

closed 2020-11-18 by junhuanchen