sipeed/MaixPy-v1_scripts #66: receiving error message : OSError: [Errno 116] ETIMEDOUT

ChahidTALHA, 2020-08-24:
Hi everyone, 
I am using a Maixduino board flashed with its latest firmware maixpy_v0.5.0_98_g7ec09ea.bin 
I was trying to implement the same code as: https://github.com/sipeed/MaixPy_scripts/blob/master/network/demo_esp32_socket.py
in order to receive an image from the server ( which is Alice.jpg) and display it on my LCD.
while doing so, i get this error message: **OSError: [Errno 116] ETIMEDOUT** in the line **60: data = sock.recv(4096)**
I successfully made a connection on my Wifi by changing the WIFI_SSID and WIFI_PASSWD.

ChahidTALHA, 2020-08-26:
Can I get a guidance about this issue ? I kept looking for a solution on the internet and couldn't find anything, if you need more information that I can provide you can ask me.
Thanks in advance.

Neutree, 2020-08-26:
make sure esp32's firmware is the latest, and if you use the master's firmware, you should change anothor resources instead of Alice.jpg, because it now become a `https` url, you can try a `http` url

ChahidTALHA, 2020-08-26:
Hi, 
The esp32's firmware version i have is **1.4.0** which is the latest.
I tried to download another resource instead of Alice.jpg to prevent the https url, to do so i tried this url : **http://via.placeholder.com/200.jpg**. This is a jpg image and has 200*200 in size
Unfortunately i still get the same error **OSError: [Errno 116] ETIMEDOUT**