sipeed/MaixPy-v1_scripts #67: receiving image from an HTTPS url

ChahidTALHA, 2020-08-27:
Hi everyone,
I am using Maixduino Board (Firmware: **maixpy_v0.5.0_98_g7ec09ea**) with its ESP32 (**ESP32_SPI firmware version: 1.4.0** ) to implement a project,
I referred to the code in the link : 
https://github.com/sipeed/MaixPy_scripts/blob/master/network/demo_esp32_socket.py
in order to download an image from **http url** and worked fine.
Now I am asking if there is any example and methods to download images from **https url** for the ESP32.
Let me know if you need more information.
Thanks in advance.

ChahidTALHA, 2020-08-31:
can i have an answer about this question?

Neutree, 2020-09-04:
try the latest  firmware, it support https now

ChahidTALHA, 2020-09-04:
Thanks for your answer,
i will check it and get back to you with a feedback.

junhuanchen, 2020-11-18:
see this https://github.com/sipeed/MaixPy_scripts/blob/master/network/demo_https_get.py

closed 2020-11-18 by junhuanchen