sipeed/MaixPy-v1_scripts #74: The ignorebutton counter waits a bit after a button

AminovE99, 2020-09-21:
## Description of the issue
When I try to  upload this code on my maixduino I receive this Traceback in console:
```
Traceback (most recent call last):
  File "boot.py", line 97, in <module>
IndexError: list index out of range
init i2c2

[MAIXPY]: find gc3028
encrypted model
[MAIXPY]kpu: img w=320,h=240, but model w=128,h=128

Traceback (most recent call last):
  File "main.py", line 18, in <module>
ValueError: [MAIXPY]kpu: check img format err!
```

I created pull request to fix it, but I suppose it is not perfect solution (https://github.com/sipeed/MaixPy_scripts/pull/73)

AminovE99, 2020-09-21:
One of the cause of the error - contacts bounce of the button.
Also when button pressed more than 10 times, there is "out of range" error, because at next press (11th) we will have lenght of "record_ftrs" = 11, but lenght of "names" is only 10.

Neutree, 2020-09-25:
add debounce and add limitation will be ok

```
if start_processing:
                record_ftr = feature
                record_ftrs.append(record_ftr)
                start_processing = False
```
limit record_ftrs length according to the names' length here,
and we can also remember last record time, if time interval too short, just ignore and only set `start_processing` to `False`