sipeed/MaixPy-v1_scripts #83: gimball example script killed my m1n im confused how

teletubbiesarenotforpendingpandas, 2020-10-21:
i recently setup the gimbal example found in your demo section hear:https://github.com/sipeed/MaixPy_scripts/tree/master/demo/gimbal

i have it setup with two servos powered externally and grounded to the m1n,i ran the script and uploaded it to start at boot.
i have caps on the servos and conected the two signal wires to some pins on the m1n adaptor.
before all that i did flash the face_model_at_0x300000.kfpkg .
and it worked for about 20 seconds, pitch and roll were fine and tracking was good,but then it stoped out of nowhere and i no longer have a working board,as im an idiot i tried to set it up with another m1n board and i got the same result a dead board,
whats happend?

did i do something i shouldnt have?

i have since tested/debbuged my diy pcb shield with an arduino and all servos work fine and my arduino is still alive,so is there some reason the m1n board died?
could someone help me understand what went wrong?

i only bought these boards to test this exact funtion so i'm pritty unhappy and sad that i may need to buy another but dont want to untill i find out what went wrong?

to confirm i no longer have any serial comunication and the board no longer boots up.the only thing ive done is run this script with servo signal wires and ground connected.

i do have a maix bit but im not looking forward to breaking another board again.

parsec3d, 2020-10-24:
Yikes I was about to start building using that example... and I´ve got a maix duino and a m1n exactly for that ... :(
have you got any answer from sipeed?

teletubbiesarenotforpendingpandas, 2020-10-24:
yep,yikes i also bought them for this purpose,i did send them a message with no reply yet,
i have since tried it with maix bit with the same pcb and it works with no problems.so im still confused.
i used pins 33 for pitch and pins 19 for roll,grounded the external servo power with the m1n board and thats it,if you happy to risk the test let me know how it goes.

i would like a schematic but cannot find one i dont think they offer one yet either.
i do believe it will work fine on the maxiduino as ive seen someone else runnning it with that board but not the m1n.

let me know if you get it working on the m1n as i like these boards.

parsec3d, 2020-10-24:
Thanks! Will try with the Maixduino first to be on the safe side for now, pls let us know if you get an answer or a clue of what could have gone wrong, will post over here as soon as I get a chance to test with the maixduino.

closed 2020-11-17 by teletubbiesarenotforpendingpandas