sipeed/MaixPy-v1_scripts #87: Description of the results of speech recognizer (isolated_word)?

myuunosuke, 2020-12-25:
I was be able to register the words for speech recognition and get the results.
I see that the first value in the data set is representing the index of detected word.

(0, 484, 35, 83)  ===> detected data(0) 
(6, 392, 73, 42)  ===> detected data(6) 
(2, 409, 96, 106) ===> detected data(2) 
(6, 195, 73, 42) ===> detected data(6) 

What are the rest of those 3 values representing???
Is there any values that represents its confidence level of the result?

It seemed like the 2nd value goes up when the confidence level is low..., 
but I couldn't find any documentation to validate that...

Thank you,

junhuanchen, 2021-01-08:
https://github.com/sipeed/MaixPy/blob/d78d036a2a74854974319d58f34e571615ec7d91/components/micropython/port/src/speech/isolated_word/py_isolated_word.c#L202-L215

see this.

myuunosuke, 2021-01-21:
Thank you, I will look into it.

closed 2021-01-21 by myuunosuke