sipeed/MaixPy-v1_scripts #90: 分类训练结果标签顺序出错，标签文本格式建议修改！

embedream, 2021-01-20:
尝试MaixHub上的模型训练，简单用了三个模型，按教程用MaixBit采集图片，上传训练，一起顺利，只是遇到点小困惑：其中两个标签交换了，确定不是我图片文件夹名称错所致。

在尝试修改标签时，发现所提供的程序根本没有使用输出结果中的标签文件 Labels.txt

尝试修改程序，发现标签文档格式不太合理，原始输出的内容是：
labels = ["SDCard", "cover", "pencil"]

这个导入到程序中，转换为 list 有点麻烦。

建议将 labels.txt 的内容修改为：
["SDCard", "cover", "pencil"]

这样就可以使用 json.loads(str) 方式导入了。

修改后的程序附在下面，供学习者参考：

# object classifier boot.py
# generated by maixhub.com

import sensor, image, lcd, time
import KPU as kpu
import gc, sys
import ujson

def lcd_show_except(e):
    import uio
    err_str = uio.StringIO()
    sys.print_exception(e, err_str)
    err_str = err_str.getvalue()
    img = image.Image(size=(224,224))
    img.draw_string(0, 10, err_str, scale=1, color=(0xff,0x00,0x00))
    lcd.display(img)

def main(labels = None, model_addr="/sd/m.kmodel", sensor_window=(224, 224), lcd_rotation=0, sensor_hmirror=True, sensor_vflip=True):
    sensor.reset()
    sensor.set_pixformat(sensor.RGB565)
    sensor.set_framesize(sensor.QVGA)
    sensor.set_windowing(sensor_window)
    sensor.set_hmirror(sensor_hmirror)
    sensor.set_vflip(sensor_vflip)
    sensor.run(1)

    lcd.init(type=1)
    lcd.rotation(lcd_rotation)
    lcd.clear(lcd.WHITE)

    if not labels:
        try:
            with open('labels.txt','r') as f:
                labelstr = f.read()
                print(labelstr)
                labels = ujson.loads(labelstr)
        except Exception as e:
            print("no labels.txt")
            img = image.Image(size=(320, 240))
            img.draw_string(90, 110, "no labels.txt", color=(255, 0, 0), scale=2)
            lcd.display(img)
            return 1

    try:
        img = image.Image("startup.jpg")
        lcd.display(img)
    except Exception:
        img = image.Image(size=(320, 240))
        img.draw_string(90, 110, "loading model...", color=(255, 255, 255), scale=2)
        lcd.display(img)

    try:
        task = None
        task = kpu.load(model_addr)
        while(True):
            img = sensor.snapshot()
            t = time.ticks_ms()
            fmap = kpu.forward(task, img)
            t = time.ticks_ms() - t
            plist=fmap[:]
            pmax=max(plist)
            max_index=plist.index(pmax)
            img.draw_string(0,0, "%.2f : %s" %(pmax, labels[max_index].strip()), scale=2, color=(255, 0, 0))
            img.draw_string(0, 200, "t:%dms" %(t), scale=2, color=(255, 0, 0))
            lcd.display(img)
    except Exception as e:
        raise e
    finally:
        if not task is None:
            kpu.deinit(task)

if __name__ == "__main__":
    try:
        labels = ["SDCard", "cover", "pencil"]
        # main(labels=labels, model_addr=0x300000)
        # main(labels=labels, model_addr="/sd/m.kmodel")
        main(model_addr="/sd/m.kmodel")
    except Exception as e:
        sys.print_exception(e)
        lcd_show_except(e)
    finally:
        gc.collect()

Neutree, 2021-01-21:
标签不需要导入， 在 boot.py 中 已经写入了 labels，  labels 文件可以不用理会， 另外可以直接用 exec() 函数执行文件内容也是可以的

Neutree, 2021-01-21:
标签顺序出错 是什么现象， 就是 boot.py 中的 labels 和 文件 labels.txt 的内容顺序不同么

embedream, 2021-01-21:
我一共建立了三个对象：SDCard 、cover、pencil
运行程序后，识别结果（LCD上显示的）cover 和 pencil 交换了。
我特地检查了采集的图片目录名，和内容对应，所以判断是训练输出的结果出错了。

boot.py 中的顺序和 txt 文件一样。

embedream, 2021-01-21:
谢谢！

我是初学 Python，还有很多不会，原来是搞 MCU 编程的，接触C语言多一些。

按提示尝试了，似乎还有什么地方没搞清楚。

按提供的原始程序修改如下：

if not labels: 
        try:            with open('/sd/labels.txt','r') as f:                 print(f.read())                 exec(f.read())                 print(labels)            #with open('labels.txt','r') as f:            #    labelstr = f.read()            #    print(labelstr)            #    labels = ujson.loads(labelstr)

运行后结果为：
labels = ["SDCard", "cover", "pencil"] 
None
 

好像 labels = 。。。 没有正确执行，不知道是哪里写错了

盼指点，谢谢

embedream, 2021-01-27:
搜了若干资料，还是没有搞定 exec 函数，尝试了locals（），也不得要领，不知道错误在哪里。