sipeed/MaixPy-v1_scripts #94: Error with Kmodel Version

Bashdog, 2021-01-23:
I can't upload the demo_find_face.py script. I just got the error: ValueError: [MAIXPY]kpu: load error:2002, ERR_KMODEL_VERSION: only support kmodel V3/V4 now, but I flashed the  maixpy_v0.6.2_16_gaefa3193d_openmv_kmodel_v4_with_ide_support.bin file to the MaixBit. 
Is something wrong with my device? Could anybody help me?

ianferreira, 2021-02-01:
I have the same issue, using the script above with firmware 0.6.2.  is the v4 model loaded on another address?

ianferreira, 2021-02-02:
It looks like the firmware doesn't have the right model embedded. The demo worked if you load http://dl.sipeed.com/MAIX/MaixPy/model/face_model_at_0x300000.kfpkg with the KFlash tool. @Bashdog

Bashdog, 2021-03-18:
Ok thank you!

rsingh2083, 2021-08-01:
Where to load this ?? Ive already loaded :- 
FaceDetection (0X300000), 
FaceExtraction(0X500000), 
FaceLandmarkDetection(0X400000) 

and then I loaded :-
maixpy_v0.6.2_57_gae955b706_openmv_kmodel_v4_with_ide_support.bin  at (0X0000)

> It looks like the firmware doesn't have the right model embedded. The demo worked if you load http://dl.sipeed.com/MAIX/MaixPy/model/face_model_at_0x300000.kfpkg with the KFlash tool. @Bashdog

rsingh2083, 2021-08-01:
> It looks like the firmware doesn't have the right model embedded. The demo worked if you load http://dl.sipeed.com/MAIX/MaixPy/model/face_model_at_0x300000.kfpkg with the KFlash tool. @Bashdog

After loading all the files, I flashed your .kpfg still face_recognition giving error about KMODEL

desaikavan03, 2024-09-20:
i m facing same issue what should i do now ??