sipeed/MaixPy-v1_scripts #95: ImportError SpeechRecognizer

deby13, 2021-01-24:
I am trying many examples for Maixduino and all of them does not work but this I would like see it working which is the Speech recognition, it show the following error, ImportError cannot import name SpeechRecognizer, I've been searching all around and could not find this module, How can I fix this, please advise

deby13, 2021-01-24:
just adding an info, I have tried with MAix bit and MAixduino

myuunosuke, 2021-01-30:
"SpeechRecognizer" has been changed to "speech_recognizer".

Please see this thread.
https://github.com/sipeed/MaixPy/issues/368

Try "help('modules')" command and make sure your firmware is built with speech_recognizer module.
If not, you will need to build your own firmware with "speech_recognizer" enabled.

I've tested  "isolated_word.py" with my Maixduino with firmware v0.6.2 with speech_recognizer enabled, and it worked.

closed 2021-02-03 by deby13