sipeed/MaixPy-v1_scripts #101: OSError: [Errno 12] ENOMEM trying run self_learning_classifier_load.py

deby13, 2021-02-23:
I have tried uploading full, minimum and minimum with k4 firmware. Also was able to upload the smodel on SD after many tries.  Then I run  self_learning_classifier.py with success. But  when trying to run self_learning_classifier_load.py I get  OSError: [Errno 12] ENOMEM. Any help would be very much appreciated

lyt-cloud, 2022-04-30:
Its original routine save statement is not written to the loop, you can add the save statement to the loop, and then load

lyt-cloud, 2022-04-30:
> I have tried uploading full, minimum and minimum with k4 firmware. Also was able to upload the smodel on SD after many tries. Then I run self_learning_classifier.py with success. But when trying to [...]

Its original routine save statement is not written to the loop, you can add the save statement to the loop, and then load