sipeed/MaixPy-v1_scripts #106: network_wiznet5k.py SPI id

jay6621, 2021-03-24:
The link [https://maixpy.sipeed.com/en/libs/machine/spi.html#parameters](url) clearly states that 'id' for SPI allowed is 0 to 3 but example here under Network folder -  network_wiznet5k.py has id: 4 , How what is the right data ? I would love to help writing or developing Ethernet support for Maix Dock. Any Help is appreciate.

H3adcra5h, 2021-03-26:
4 is the same as SPI.SPI_SOFT

jay6621, 2021-03-27:
So, it is using software SPI ? How about using hardware SPI ? I couldn't find any pinout diagram for Maix Dock.

H3adcra5h, 2021-03-28:
The documentation isn't so clear on this side.
You have 4 hw SPI devices.
SPI2 can only work in slave mode, but I read somewhere. maybe in the documentation, that the slave mode isn't implemented.
SPI3 is reserved the the flash.
SPI0 is the lcd, I think and
SPI1 is the SD card, but also not sure. If I use SPI1 the SD card isn't available.

So there is no free hw SPI device for the network.
Some interesting material can be found [here](https://cn.dl.sipeed.com/shareURL/MAIX/HDK/Sipeed-M1&M1W).