sipeed/MaixPy-v1_scripts #108: not able access bind() for socket communication.

AMZzee, 2021-03-26:
I am trying to create a tcp server from my maix dock m1w, inorder to communicate with my android app(i.e the client). To do so i need to bind my host ip and port address to the socket.

When i run:

`import socket`
`s=socket.socket()`
`s.bind((ip_addr,portno))`

i get the following error:
`AttributeError: 'socket' object has no attribute 'bind'`

Can someone please tell me, why there's no bind function in the firmware? or am i going wrong somewhere? Would appreciate any help.
Thank you.

Please complete the following information

IDE version: 0.2.5
Firmware version: maixpy_v0.6.2_32_gf1cc361a5_openmv_kmodel_v4_with_ide_support
Board: Maix Dock m1w
OS: Windows 10

H3adcra5h, 2021-03-26:
No, you are right, it's not implemented see: [https://maixpy.sipeed.com/maixpy/en/api_reference/standard/usocket.html](https://maixpy.sipeed.com/maixpy/en/api_reference/standard/usocket.html). 
Why not is a very good question. I think too much work to implement the whole at command set.  The network support is very basic and the esp-at firmware in the esp8285 is very, very old.

AMZzee, 2021-03-27:
Hey, thanks for the reply. So, can you tell me if there's any way i can make my maix dock a tcp server?

H3adcra5h, 2021-03-27:
No way yet. The only way is writing your own esp-at library, but it's a lot of work.