sipeed/MaixPy-v1_scripts #118: Sipeed Maixpy Amigo and Cube, on board mic don't work

YvesLafont, 2021-11-10:
Hi,
i'am unable to get the internal microphone working on Amigo and Cube.
Have someone a working MaixPy source to check the internal mic ?
I tried several exemples, checking the correct I2S mapping pins, but no result.
Compiling, is ok, without runtime errors, but no result
I use the last 6.2 firmware (normal, not minimum)
Thank you very much for help.

Yves

YvesLafont, 2021-11-10:
Update :  Internal microphone work well when uploading "es8374.py" and care with i2c and i2s pins. The FFT demo work
But that work not for the Cube .. The error is "pins 33, 34 and 35 are used by unknown" (I2S Cube pins for receiver)
What is the correct definition for the Cube ?

junhuanchen, 2022-01-26:
https://wiki.sipeed.com/soft/maixpy/en/course/media/audio.html

closed 2022-01-26 by junhuanchen