sipeed/MaixPy-v1_scripts #121: Audio record/play example for MaixCube?

junfengzh, 2022-01-25:
Do we have an example for recording and playing audio on MaixCube? I tried https://github.com/sipeed/MaixPy_scripts/blob/master/multimedia/audio/record_play.py but it does not work.

junhuanchen, 2022-01-25:
在提 issue 前能先看文档 wiki.sipeed.com 吗？

closed 2022-01-25 by junhuanchen

junfengzh, 2022-01-25:
Can you be more specific and link the sample code or config for MaixCube? I looked at wiki.sipeed.com but have no idea. Help appreiciated.

junhuanchen, 2022-01-26:
https://wiki.sipeed.com/soft/maixpy/en/course/media/audio.html

junhuanchen, 2022-01-26:
https://wiki.sipeed.com/soft/maixpy/zh/course/media/audio.html

junfengzh, 2022-01-26:
谢谢回复。我正是按照这些例程来试的。如下是我的代码，和record_wav.py一模一样只是去掉了写文件环节。
打印出来的音频数据全是0。我怀疑是GPIO的端口和MaixCube的不一致？

```
import image, lcd, time
import audio
from Maix import GPIO, I2S
from fpioa_manager import fm

# user setting
sample_rate   = 16000
record_time   = 4  #s
# default seting
sample_points = 2048
wav_ch        = 2

fm.register(8,  fm.fpioa.GPIO0, force=True)
wifi_en = GPIO(GPIO.GPIO0, GPIO.OUT)
wifi_en.value(0)

fm.register(20,fm.fpioa.I2S0_IN_D0, force=True)
fm.register(19,fm.fpioa.I2S0_WS, force=True)    # 19 on Go Board and Bit(new version)
fm.register(18,fm.fpioa.I2S0_SCLK, force=True)  # 18 on Go Board and Bit(new version)

rx = I2S(I2S.DEVICE_0)
rx.channel_config(rx.CHANNEL_0, rx.RECEIVER, align_mode=I2S.STANDARD_MODE)
rx.set_sample_rate(sample_rate)
print(rx)

frame_cnt = record_time*sample_rate//sample_points

for i in range(frame_cnt):
    tmp = rx.record(sample_points*wav_ch)
    print(tmp.to_bytes())
```

junhuanchen, 2022-01-26:
[image]

junhuanchen, 2022-01-26:
[image]

junfengzh, 2022-01-28:
https://github.com/sipeed/MaixPy_scripts/blob/master/modules/others/es8374/es8374.py 能用，多谢。