sipeed/MaixPy-v1_scripts #127: No module named ‘speech_recognition’

renanmbs, 2022-08-31:
Hello,

I am trying to upload the speech recognizer - isolated word on my maix dock m1w. Every time it gives me the following error:

_[ModuleNotFoundError: No module named ‘speech_recognition’]_

I did pip (and pip3) install SpeechRecognition on my mac terminal, and it installed. Now it says:

_Requirement already satisfied: SpeechRecognition in ./opt/anaconda3/lib/python3.8/site-packages (3.8.1)_

I upload the script _isolated-word_ from MaixPy Scripts (again) into my MaixPy IDE, but it still gives me that error.

What do I do to solve that?

Neutree, 2022-08-31:
maybe you should download this firmware
[image]

renanmbs, 2022-08-31:
I’ll upload the speech one. Also, my system is on firmware 5. To update it, do I download the last one from the list? (G8…58.bin)

Last time I tried updating the firmware, it gave me an error on the register to write it on. Do I just leave it on 0x00000?

Neutree, 2022-08-31:
firmware should write to address 0x00000 of flash